martin2250/GrblHeightProbe2 #2: macos binary?

beckmx, 2015-05-28:
Hello, is there any chance to have a macosx binary?

martin2250, 2015-05-31:
Hi, I don't know if it will work, but you could try to use Mono to run the normal executable on Mac/Linux.
I don't know of any way to compile C# code to a native MacOS binary.
If you run into any problems let me know, I will test it on Linux tomorrow.

beckmx, 2015-05-31:
I alredy tried to run it on mono, it does open without problems, but the
ports are not listed correctly and it doesn't recognize the correct port,
and well it doesn't allow me to drag&drop the gcode into the window

beckmx, 2015-06-01:
Ok I am running it using my VMWare machine, but I notice something, each time my grbl board connects it looks it resets, so when using the probe application the commands do not get to the board, I think adding a $X or "unlock" button should be ok, and also a reset zero button.

martin2250, 2015-06-01:
Thanks for the heads up with the port listing, that is certainly something fix-worthy. Other than that, the program isn't supposed to do drag'n'drop, that was an older version of the program (If you want that feature,  I could add it though)
I don't think extra buttons for $X or resetting zero are necessary, you can just enter them into the console. (G92X0Y0Z0 for resetting zero)

martin2250, 2015-06-01:
I added a way to enter port names manually, it's only in the dev branch so here is a binary to try it out: https://www.dropbox.com/s/n1k83mkzbyavj5f/GrblHeightProbe2_mono_experiment.exe?dl=0  
on ubuntu, I was able to open the port and my machine would blink a bit from being reset, but neither sending nor receiving seem to work. Let me know how it works on mac

beckmx, 2015-06-01:
I will test it in 3 hours, when I hit lunch time ;)

sparkiedave123, 2015-10-23:
I compiled using Microsoft Visual studio 2015 on a windows 7 computer.
It compiles ok but I do not get the drop down menu item "manual console".
It also crashes when I click to start the probing.
Is this an issue with your program or the way I am compiling it ?
Could you send me the .EXE ?

martin2250, 2015-10-24:
I addressed your problems in a comment on the other issue.
You can download a precompiled binary here: https://github.com/martin2250/GrblHeightProbe2/releases, it is up-to-date with the latest commits.

sparkiedave123, 2015-10-24:
Hope this helps

System.FormatException was unhandled
Message: An unhandled exception of type 'System.FormatException' occurred in mscorlib.dll
Additional information: Input string was not in a correct format.

From: martin2250 
Sent: Saturday, October 24, 2015 11:37 AM
To: martin2250/GrblHeightProbe2 
Cc: sparkiedave123 
Subject: Re: [GrblHeightProbe2] macos binary? (#2)

I addressed your problems in a comment on the other issue.
You can download a precompiled binary here: https://github.com/martin2250/GrblHeightProbe2/releases, it is up-to-date with the latest commits.