martin2250/GrblHeightProbe2 #6: Documentation

tokyis, 2015-07-27:
Hi,
I'm looking into pcb milling and thought your app would help greatly but I found myself a bit surprised when my xcarve slammed the bit in the FR4 and made a diagonal to the center of the board, I can still cut the board into brackets and salvage it for solder paste stencil applications but I only have so many engraving bits so I'd like to know if I did anything wrong and fix it.
I'll describe what I did and what I was expecting to happen:

-Clipped the spindle with the clip I use for probing
-Clamped the board I use for Z probing/zeroing conductive face down on the FR4 board

-Using UGS:
  -Homed the machine
  -Jogged to a position inside the board that is 3mm away from the edge and roughly 6mm above
-Closed UGS

-Opened GHP2:
-in settings I set Probe Plunge to 1 and Probe depth to -5 (spider sense tingling maybe)
-In the console I issued a $x
-defined a 20*20 5mm grid with 0*0 offset
-Started probing
-Z axis went straight down about 2mm under the surface, Z axis skipping steps (I assume it would have gone through the FR4 if my stepper currents were set higher)
-Saw a probe fail briefly and the machine went in rapid to the center of the area I defined, gouging the board, followed by an alarm lock.

I was expecting the machine to rapid down 1mm, probe down to -5mm from that position, rinse, repeat.

I tested a few probing commands in UGS after that and it works fine.
Thanks in advance for light you can shine on this ^^

tokyis, 2015-07-28:
I tried again with the board balanced on something so that if it went too far or moved while down I would have time to hit the Estop and it turns out on my first attempt I did not set a WCS as I was assuming it moved relative to its starting position, that was dumb of me.
My previous wcs was set for a 2mm aluminium plate with a shorter endmill, which is why it slammed into the board...
It would be nice to have a checklist in the doc., of things that MUST be setup before launching the probing, so nothing is randomly assumed. ^^
Also the Probe Plunge setting is a little odd, I first assumed it was an initial move down, but it's a feedrate, IMO it should say something like "Probe plunging feed" or "Probing feedrate".
While I'm on suggestions, the console could be non modal or integrated to the main UI since you always need it, it would save the motion of going into a menu, just disabling the buttons and input field when the app is probing.

Sorry for the useless bother, hoping it helps avoid other wrecks. ^^

closed 2015-07-28 by tokyis

martin2250, 2015-07-28:
Thanks for the input!

I could integrate the console into the main window, like UGS does.
Maybe I'll add a real-time position readout to the bottom of the screen, this should also prevent any coordinate system errors.

Also concerning your edit: why is it a bad thing to immediately go to retraction height? The entire probe area should be free of clamps anyways.

Greetings

tokyis, 2015-07-30:
Thanks for your reply,
My thinking was that you might not be over the probing area when starting, you might be home or have sent the spindle in a corner to do some cleanup before probing...
Going within the probing area from the safety height before going down further means that unless the user set the safety wrong you cannot hit anything outside the probing area.
IMO it covers more scenarios that way...(I use small squares of double sided tape under the board and at least one clamp to hold the ground plate on the board for the probing and it sticks out 13mm, I just know I'm gonna forget to G30 above the probing area one day ^^)