martin2250/GrblHeightProbe2 #7: Path not contained in heightMap

DennisDG, 2015-09-20:
As the title say's the (Path not contained in heightMap)  so i can't convert my g-code files.

the button (apply height map and save) has become grey so i can't click on it.

How can i fix this?

martin2250, 2015-09-20:
Does the path actually fit within the area of the height map? Not only the size but also the absolute position matters. You can use the translate tool to move the tooppath within the area that you probed.

DennisDG, 2015-09-21:
Yes it works, the problem was that the code was negative instead of positive

martin2250, 2015-09-21:
Great! Aside from that, how was your first impression of the program?

closed 2015-09-21 by martin2250

DennisDG, 2015-09-21:
It is a realy great interface. 
But do you know how I can put the height map program into your own stlye.
 
maybe with microsoft visual basic's ? 
 
Thank you for helping.
 
Dennis de Graaf
 
Date: Mon, 21 Sep 2015 03:53:18 -0700

martin2250, 2015-09-21:
what exactly do you mean by "put the height map program into your own stlye"?

DennisDG, 2015-09-21:
Euhm I mean, is there a way that you can make your own layout or make one program including grbl controller and height map?

 

 

-          One very important thing and easy to edit as a programmer:        When you remake a gcode with the GrblheightProbe2 program, the depth has changed  from -0.2mm to -1,5mm.

So if I start with engraving, my router puts the milling bit almost fully trough my PCB board. If you want to make it a good trace you can do the next thing   -1.5 - -0.2 = -1.3    so you have to zero your router +1.3mm above the PCB surface if you want a depth of -0.2mm.

 

Can’t you make the subtraction in the program?

It would be a lot easier for the operator.

 

 

Sorry for my bad English.

 

 

Thank you for helping me out,

 

Dennis de Graaf

martin2250, 2015-09-21:
I thought about that, but in the curent state of the program it would only result in a big mess. If I ever write a dedicated replacement for UniversalGcodeSender I could include it.  
  
Regarding your problem with depth: I never had any similar issues, might be that you simply didn't zero the Z-axis before starting to probe (manually enter "G38.2 Z-5 F25", then "G92 X0 Y0 Z0"). Check your height map by hovering over the points, it will probably have the -1.3mm offset everywhere. Remember that the machine origin you set before probing (with G92) must be identical to the one you use to route the board.  
  
 I really can't imagine what else would cause this to happen, If the problem persists please open a new Issue and attach the files that you used (height maps + gcode).