martin2250/GrblHeightProbe2 #10: "Apply heightmap and save" gives crazy output

Envergure, 2015-09-30:
I have a random little Gcode file I'm using for tests.  It was generated in MakerCAM.  When I apply a heightmap to it using GrblHeightProbe v2.3, the output toolpath is quite different from the input.
[image]
The original toolpath (left) looks correct and I've successfully carved the part already.  Applying the heightmap causes the curved sections of the path to become misshapen (centre).  I haven't yet tried sending this output file to my machine.  UGS disagrees with GHP2 as to the diameter of some of the arcs (right).

I tried to attach the heightmap and Gcode files to this post, but GitHub won't accept them :(

martin2250, 2015-09-30:
MakerCAM often produces rather bad G-Code, it uses many subsections for arcs and sometimes the center doesn't match up completely with the initial and final position.  
Can you try using version 2.2? I've made some changes to the GCode parser. You can upload the GCode to pastebin (even anonymously) or as a GitHub gist and just link it in here.

Envergure, 2015-09-30:
Version 2.2 gives the same result.

Here are the files:  https://gist.github.com/Envergure/17b4d43f75ccfddb8a48

I'll try starting from scratch with a different CAM program.

Envergure, 2015-10-01:
I've tried again, this time using gcode generated by the GCodeTools extension for Inkscape.  The result was similar.
[image]

martin2250, 2015-10-01:
Thanks for the confirmation, I'll look into it

martin2250, 2015-10-01:
I had a mistake in a single line which caused arcs to be messed up, I'm just uploading release v2.3.1 which should work properly now.

Envergure, 2015-10-01:
v2.3.1 is working perfectly on all the gcode files I tried before, both MakerCAM and GCodeTools.  Thanks so much!

martin2250, 2015-10-01:
Great to hear that, I don't know how this even made it through so many versions without being noticed. I guess I only ever used it for pcb-gcode generated files which only use straight lines, but I remeber testing it with a pretty big makercam file as well.

closed 2015-10-01 by martin2250