martin2250/GrblHeightProbe2 #13: margins extrapolation

eliasg13, 2015-10-07:
Hi!
    I'm no shure if this is the place to make sugestions, so, if i'm in the wrong place, please forgive me.

I'm milling a circuit that ocupies almost all the pcb, and it's dificult to probe the entire surface aboiding to fall of the edge. The program wont let me aply the corrections to a path that is 0.5mm bigger than the probed area. Is there any chance that you may consider to implement a way to fascilitate this?

    Thanks!!!

martin2250, 2015-10-07:
Hi elias,  
You're at the right place here!  
  
It would be possible, but normally you should be able to get around that problem by scaling your area accordingly (you can use fractions as grid size).  
I'm currently writing a completely new program that works more like UGS but still includes Autolevelling, If it really is too difficult to scale the area, I'll look into implementing it there.

eliasg13, 2015-10-07:
I saw  that you are writing it, that's awesome! It is difficult, last night
had milled the PCB and it was really really better, exept in the margin,
where it was too deep (because the triangular bit that used to probe fell a
bit from the edge)

It's a great tool! My circuit still is unusable, but probably because mine
is too ambitious. Have to go back to Cad.