martin2250/GrblHeightProbe2 #14: GrblHeightProbe2

sparkiedave123, 2015-10-24:
After compiling on windows 7 the command dropdown menu does not appear,
then program crashes when you hit the Start/resume probing button.

martin2250, 2015-10-24:
I removed the command / manual console entry and moved the console to the lower half of the main window.
As for the crash, can you run it with debugging in VS and post the exact exception?

sparkiedave123, 2015-10-24:
Ok i understand now.

This is how i use your program.

Windows 7
Arduino mega with attached script

1/ open program (your latest .exe version)
2/ Connection , open com5 (coms now established).
3/ HeightMap, new, create
4/ >Start/resume Probing.
5 / program crashes (<Idle>) shown in command window
6/ see attached image.

From: martin2250 
Sent: Saturday, October 24, 2015 11:36 AM
To: martin2250/GrblHeightProbe2 
Cc: sparkiedave123 
Subject: Re: [GrblHeightProbe2] GrblHeightProbe2 (#14)

I removed the command / manual console entry and moved the console to the lower half of the main window.
As for the crash, can you run it with debugging in VS and post the exact exception?

martin2250, 2015-10-24:
There is no attached image. Can you tell me in which line the format exception occured and maybe which string it received? That should help a lot. Also, which version of grbl are you using?

sparkiedave123, 2015-10-25:
Sorry for the confusion.
My mistake for not following your u tube video.
working fine.
Thank you for your brill program

From: martin2250 
Sent: Saturday, October 24, 2015 10:08 PM
To: martin2250/GrblHeightProbe2 
Cc: sparkiedave123 
Subject: Re: [GrblHeightProbe2] GrblHeightProbe2 (#14)

There is no attached image. Can you tell me in which line the format exception occured and maybe which string it received? That should help a lot. Also, which version of grbl are you using?

martin2250, 2015-10-26:
Great that it's working again! Can you tell me what the problem was? Maybe I can warn people about it when I finally start the wiki.

closed 2015-10-26 by martin2250

sparkiedave123, 2015-10-26:
It was just my silly mistake not following the correct procedures in the program.
Nothing wrong with it.
Thanks for a great program.
One suggestion , could you include  a machine controller i.e. X,Y,Z  in steps and jog so that you can move the axis without having to type in the G codes manually.

From: martin2250 
Sent: Monday, October 26, 2015 3:50 PM
To: martin2250/GrblHeightProbe2 
Cc: sparkiedave123 
Subject: Re: [GrblHeightProbe2] GrblHeightProbe2 (#14)

Great that it's working again! Can you tell me what the problem was? Maybe I can warn people about it when I finally start the wiki.

martin2250, 2015-10-27:
Thanks for the suggestion, but I don't think I'll include a jogging feature.
When I have some free time again, I'll continue working on my other project which will integrate autolevelling in a mix-up of EMC/MACH and UGS