martin2250/GrblHeightProbe2 #15: Program Error

ThiagoCh, 2015-11-19:
Hi, this program stops working and windows 10 closes it automaticaly every time I try to conect through COM3. Thanks :)

martin2250, 2015-11-21:
Hi,

this is a pretty unique issue so far. I only had bluescreens when two threads write to the serial port simultaneously, though I remember fixing this problem even before the first public release.

Does it crash immediately or only after some time? 
What USB to serial adapter are you using/which drivers?

Let's get this sorted out,

Greetings!

ThiagoCh, 2015-11-22:
Hi, thanks for you reply.
I'm sorry, it was just a mistake that I made. My grbl was the old one, it works with baud 9600 a day I was trying to use your software with 115200. When I changed the software to 9600 it worked. Thanks for you attention and software. Have a nice weekend.

Enviado do meu iPhone

martin2250, 2015-11-22:
Glad to hear that!

closed 2015-11-22 by martin2250