martin2250/GrblHeightProbe2 #16: Dual sided copper

DennisDG, 2015-12-10:
Hi Martin, 

Its me again because I ran into a new problem (The Problem DUAL SIDED offset).

When I've made a 2 sided PCB board layout, I generate a gcode of it with the PCB-Gcode plugin in Eagle.
 
Everything works fine unless one thing, the 2 sides have a offset of ±1mm from each other.
My orgin of the pcb stays the same because I've drilled 2 holes in the pcb and than I put a pin in the PCB and my mill  so thats oke. 

When I see my dimensions of the gcode directly after the PCB-Gcode Plugin, the top and bottom are the same. And then if I do a height probe the offset is suddenly there. Do I have to make a border around my PCB BOARD ?      

The 1 sided copper PCB's works perfectly!! Thank you for writing this program. 

Thank you,

Dear Dennis de Graaf,

martin2250, 2015-12-10:
This really shouldn't happen. Did you notice any offsets when making one-sided boards?  
If the problem persists, can you send some sample files (both gcode and height maps)

Great to hear that you like it apart from that.

Have a great one,
Martin

DennisDG, 2015-12-13:
HI Martin, 

Thank you for helping me out, I've sended a winrar with all the included files using wetransfer.

Click below for screenshots etc. 

https://www.wetransfer.com/downloads/f13f3e90214d9cdd11e69f5068f080ae20151213150005/b4fb10df9085566345627118c05c1ea020151213150005/14ac8c

Thanks for your attention and time,

Dennis de Graaf

DennisDG, 2015-12-16:
NO I haven't any offset when I make one sided boards that works fine.

Can you pleas check my file's.

Happy xmas Mate!

Dennis de Graaf

martin2250, 2015-12-18:
Hi Dennis,

I checked over your files and it looks like you used the translate feature to move the toolpath of the bottom etch into positive coordinates. Did you double check all offsets? you could take a known point from the probed gcode (eg a "spot" predrill), move your bit to these coordinates and see if it matches the existing hole. Also, your machine could loose the exact position while changing tools. I usually have to re-zero my machine after switching from the V-bit to the drills.

Happy Christmas to you too, pal. Have a good one!

DennisDG, 2015-12-28:
yup everything is fine now. Thank you Mate.

closed 2015-12-28 by martin2250