martin2250/GrblHeightProbe2 #17: Version Confusion

crezyoz, 2016-01-21:
I have downloaded and built V2. But I am unable to apply the map to a file. Unlike the video, our GCode menu item only shows "open". When the GCode file loads, Apply Map to File and save, is grayed out. Please advise.

martin2250, 2016-01-21:
Is it possible that the map file doesn't cover the entire area of the board? even rapid movements must be included as well as the XY-origin. It should automatically be enabled as soon as the toolpath fits within the map. How did you generate the map and gcode?

crezyoz, 2016-01-21:
Hello Martin! Thank you for your reply. Greatly appreciated.

It is possible. Though I thought I had set the height map params to 
cover the board +1mm or so. I am assuming it is ok to generate a map 
that is larger than the board?

I am using Eagle and pcb-gcode to generate the map. At first, while 
trying to generate the map I was driving the spindle right into the PCB! 
Then I used a G38.2 to do a touch off and followed it with a G92 X0 Y0 
Z0. From then, the app did a beautiful map. If that is the possible 
reason for my problem then I will try again with a larger map. Makes 
sense.  I now realize what the "Generate Height Map' function does and 
that I don't really need it for what I am doing. The reason for the 
'double' exception was that I did not have an expression in there.

I am not sure what you mean by 'rapid movements'.

I am new to this. As an electronics developer I haven't made my own PCBs 
yet. I was thankful to find your app because I can not get chilipeppr to 
work well. I don't much like the thought of a web based app for milling 
anyway!

Thank you very much for your reply. I will go at it again today with a 
larger map and see what I can get working.

Phil

martin2250, 2016-01-21:
Hi Phil,

yes it is ok to generate a map that's bigger than the toolpath. On the bottom infostrip it should show the exact extends of the board so you can use them as a reference for your probed area.
If you don't want to waste time probing the board again without success, you can try applying a generated height map.
Rapid movements are all movements where the tool does not cut any material or travels between different cuts. The GCode Parser assumes the initial position of the machine to be X0Y0 so that's why the origin has to be within the bounds.
And just in case: don't forget to set the offset of the probed area. If you generate a toolpath for the bottom copper side, pcb-gcode moves it to negative X coordinates, so you'll need to set the offset to (-) the width of the board or use the 'translate' feature to move it into positive space.

Greetings,
Martin

crezyoz, 2016-01-21:
>
> Hi Phil,
>
> yes it is ok to generate a map that's bigger than the toolpath. On the 
> bottom infostrip it should show the exact extends of the board so you 
> can use them as a reference for your probed area.

Sounds good. I found this morning that I had to exaggerate the size a 
little from the board size.

> If you don't want to waste time probing the board again without
> success, you can try applying a generated height map.
> Rapid movements are all movements where the tool does not cut any
> material or [...]
Aha.. Ok I get the rapid movements now. My board starts at 0.0 , 0.0 so 
I have been using a zero offset. I think that is right?

> And just in case: don't forget to set the offset of the probed area. 
> If you generate a toolpath for the bottom copper side, pcb-gcode moves 
> it to negative X coordinates, so you'll need to set the offset to (-) 
> the width of the board or use the 'translate' feature to move it into 
> positive space.

Hmm.. I don't quite understand that but likely I will after I screw up a 
couple more boards!

I may have a couple of questions later but it is all becoming more 
clear. I added a couple of buttons to the app for setting zero and doing 
a single probe event.

Thanks again for your help. I hope you don't mind.

> Greetings,
> Martin

martin2250, 2016-01-22:
No worries, mate!

Just to clear things up a bit concerning offsets:
GCode for the bottom side of the PCB will start at negative X-coordinates, so it will cover a width of say X-49.5 to X0. This means you have to probe a 50mm wide area starting at X-50. So your height map parameters for the X-axis would be 6 points with an offset of -50.

Always happy to help!

crezyoz, 2016-01-22:
Thanks Martin! I am still trying to get a decent single sided board 
done! My fault mostly. Your app is working quite well. I want to thank 
you for posting it. I will be back in touch!

Phil