martin2250/GrblHeightProbe2 #19: No connection

JacobDekker, 2016-02-06:
When you choose "Connection"->"Open"->"Comx" and there is nothing connected to your computer on that COM port x, the program will crash. 
Maybe it is better to give a Warning Message?
Having said that, I think you made a great app.
I use Ms Windows 8.2
grtz, J.Dekker

martin2250, 2016-02-06:
Thanks for the feedback!
What do you mean by "nothing connected", do you mean that there's nothing attached to the serial port or that the serial adapter isn't connected to the computer? The first case shouldn't crash and just do nothing and the second case should give a message.
Greetings!

JacobDekker, 2016-02-07:
thnx for answering.
I had nothing connected to the COM port, no COMports were in use at all.
On my laptop (MSI DE 70 + W8.2) the program crashed.
I had to restart.
regards, JD

From: martin2250 
Sent: Saturday, February 6, 2016 10:30 PM
To: martin2250/GrblHeightProbe2 
Cc: JacobDekker 
Subject: Re: [GrblHeightProbe2] No connection (#19)

Thanks for the feedback!
What do you mean by "nothing connected", do you mean that there's nothing attached to the serial port or that the serial adapter isn't connected to the computer? The first case shouldn't crash and just do nothing and the second case should give a message.
Greetings!

—

Geen virus gevonden in dit bericht.
Gecontroleerd door AVG - www.avg.com
Versie: 2015.0.6176 / Virusdatabase: 4522/11572 - datum van uitgifte: 02/07/16