martin2250/GrblHeightProbe2 #23: Cant seem to apply to Gcode

tekadept, 2016-04-25:
Heya I go through and enter a new height map. 
for me i go 3x3 points. -280 -280 offset with  a grid of 130

This gives me dimensions -280 - -20mm

I go through and  start probing, it goes through and does the probing fine.. i then go open gcode, and open my file. the cutting area is 250x250 and pops up athe preview. i goto file and apply heightmap is greyed out. and in bottom right it says "Path not contained in heightmap"

I have 30mm clearance of the object so I am unsure as to why it isnt working? am i doing something wrong?

martin2250, 2016-04-27:
Since all toolpaths start in the origin, you need to make sure that (0,0) is included within your area. Using your settings, you still are 20mm shy of the origin.

Also I'm a bit curious: why only 3x3? I always used a grid size in the order of 5-10mm. Let me know how it works out

closed 2016-04-27 by martin2250

tekadept, 2016-04-27:
Heya I am not using this for PCB's, I engrave mdf & other woods, and the thckness of the material varies by .05 to .15 mm, and when you are engraving 0.1-0.2mm it makes a large difference. I am only going 3x3 across a 280mm sheet figuring that will be enough to average out the depth, once i eventually get it working i can increase if need be, just trying to avoid too much time probing each sheet.

Hmmm 0,0 im not able to iinclude at the moment.. I have a fence there to butt the stockagainst and my probe is round hence it wont fit. I might have to jury rig up a pcb board and use that as a probe so that i can get in to 0,0. Will experiment and let you know how i go .

martin2250, 2016-04-28:
well, the easiest solution would probably be to just set the WCS to your current (-20,-20) and then probe from -160 to 0. You can use the "move path" tool to position the work correctly again.