martin2250/GrblHeightProbe2 #26: Does not work on Windows 7 x64

maximius, 2016-06-01:
First of all I want to say a huge thank you for such a needed project. milling the PCB on the CNC for me at this stage in the end nepredskazuemy result due to irregularities of the Board. I think that your program could help in the manufacture of printed circuit boards. 
[image]
[image]
Downloaded Your program and exe, and master. The program grbl connects to, executes commands $X and G92, but after clicking on the start button the program closes. The program is located at C:\GrblHeightProbe2.exe. GRBL 0.9 j. on Windows XP x86 the program does not start at all. I tried to run the previous version - the same result. Help me, please.

martin2250, 2016-06-03:
Hi,
Thanks for including a screenshot of the debug message!
It looks like your grbl configuration does not report work- and machine coordinates in the status update.
I'm not sure if it is a compile time option or configurable with commands, but **make sure that grbl reports both WPos and MPos when you send a '?'**
Also, parsing floats with language settings other than english is a gamble, since some other languages use ',' as a decimal separator, but I don't think this is the problem here.

maximius, 2016-06-05:
Thank you for answering my question. I did everything you said. a screenshot of the program below. 
My steps:
1. The Height Of The Map > New > Ok
2. Connection > Open > Com 3

I tried to run the probe command g38.2 - it works well. then raised the spindle up and pressed start. The program closed once again with the same error.

maximius, 2016-06-05:
I found another mistake. I once again started debugging. after the command g38.2 grbl replied "Probe alarm" and stopped responding to other commands.
[image]

martin2250, 2016-06-05:
It really shouldn't send any Probe fails after it has reported a successful probe point, does it behave the same way if you play around with G38.2 in UGS?

maximius, 2016-06-06:
Run the command g38.2 in UGS - the same result. Everything works fine until I start the command g38.2. After that any movement is accompanied by the error "probe fail". Don't understand why because I downloaded the firmware from github. What you use a version of grbl. Please contact me on Skype to help solve this problem. My nickname maximius2009.
[image]

martin2250, 2016-06-06:
I don't have much time at the moment, so no skype probably. did you connect the wire to the probe pin right (does it react to continuity to ground)? (I doubt that it's that, but you never know) but judging from the screenshots it did react somewhat.
Can you try the grbl 0.9g binary? IIRC this is the last version that I used successfully

maximius, 2016-06-07:
Flashed version 0.9 g. The debugger crashes on the same line as before.
Noticed that the versions of grbl 0.9 g and 0.9 j different answers to the command g38.2: 0.9 g response to a command of the form [PRB:0.000,0.000,-1.388], and 0.9 j [PRB:0.000,0.000,-1.388:1]. The last parameter is the logical answer, probe successful or not.

Also noticed that the Probe fail error only appears after it has entered the command g91. Unless you use it, and use g0, then there is no error. 
Don't know what I'm doing wrong. Can to work grblheightprobe need to install any software?
[image]

martin2250, 2016-06-10:
Ah, right. I remember updating Height Probe to accept the new :1 probe successful response, sorry for the mix up. I really can't explain why this is happening, but right now it seems like a problem with grbl. You should contact the devs of grbl, I'm sure that probing will give you very good results once it's working.

maximius, 2016-06-22:
I found another program that allows you to scan the surface of the PCB. This chilipeppr. Sometimes in the console there was an error "ALARM: probe fail", but in the end in this program, I was able to scan the surface of the printed circuit Board and map. Wrote grbl developers to help with this problem.