martin2250/GrblHeightProbe2 #27: Does not want to Apply Height.

Jacoleroux, 2016-08-01:
Hi, Thank you for your software,

I'm having trouble applying the height map,

I probe an area of X 96.6 mm Y 69.0 mm, the PCB is X 95.45mm * Y 56.08mm
When I go to File , apply Height map is not highlighted for selection. 

I have tried to make the PCB X 94mm but still not working.

What can I do to fix. 

[image]

martin2250, 2016-08-02:
Hi, glad you like it!

where does your height map start? the height map's absolute position is important.
you have to set the origin to something like (-96, -1) and then make the size (97, 58).
If you still have trouble, attach a screenshot of your settings in the "Create HeightMap" dialog.

Greetings!

dennishatton, 2016-09-05:
I'm having the same problem
My area is 45mm x  45mm
The origin is 0 x 0
What do you mean path?

Jacoleroux, 2016-09-05:
Hi Dennishatton,
Well what I have seen is the following, open GrblHeightProbe2 then open you GCODE file and check the below dimensions, if it starts with - enter it in negative form .. Let me know if you got it going, sorry i'm a bit in a rush today but will check in later.