martin2250/GrblHeightProbe2 #28: GRBL configuration question

burtbick, 2016-08-13:
Hi, First off this looks like an interesting program, and I'm looking forward to trying it out.  I've gotten my probing wired up and I've been able to test with the auto-leveling that is built into bCNC so that is all working.

But when I try to run the GrblHeightProbe2 software (using mono on Linux) it connects to GRBL but doesn't do anything.  In my case I have hard limits enabled so typically when I first connect to GRBL I have to run a Homing cycle.  Since I had to connect to GRBL using bCNC to set my X/Y/Z Zero position and then disconnect so I could connect with GrblHeightProbe2 I suspect that my problem is that GRBL is expecting that I will do a $H (home).

So I wanted to see if you have your machine configured for hard limits (with limit switches) or if you have the hard limits disabled in GRBL?

Once your new integrated gcode sender with auto leveling is completed that would of course also solve the issue.

Thanks,
Burt

martin2250, 2016-08-14:
Hi Burt,

you can always home your machine in the manual console in HeightProbe.
The problem is that Mono's implementation of the SerialPort class does not include (never fires) the Data Received event which this application relies upon. Right now the only fix would be to run it on Windows.

The integrated gcode sender/leveller probably won't happen very soon, so I'd use a VM if you want to give it a try. (I haven't tried bCNC's autoleveller, so I can't tell you if it's worth the upgrade)

Greetings,
Martin

burtbick, 2016-08-15:
Thanks Martin,The bCNC auto level works well enough, but your display of the map is a bit nicer.  One advantage of the bCNC auto level is that you run the auto-level and then the map will be applied to your gcode as it runs so you don't need to generate a modified gcode file.  Of course that doesn't give you the opportunity to examine the gcode to make sure that it looks OK.I haven't played with mono lately so I wasn't aware of the issue with the SerialPort class.  If I get some time I'll dig into the mono SerialPort class implementation to see how difficult it would be to have it include the Data Received event processing.I'll also try testing using a VM with Windows running there on my CNC computer.  If that doesn't work then I'll try running it on my notebook with Windoze on it for a test.Regards,Burt

(#28)
<author@noreply.github.com>

martin2250, 2016-08-15:
If you don't have the VM up and running go for the notebook for a first test.
If you decide to use it, I'd recommend modifying [GRBL.cs](https://github.com/martin2250/GrblHeightProbe2/blob/master/GRBL.cs) instead of Mono. Just add a timer that fires a slightly modified Port_DataReceived method.

Good luck,
Martin

burtbick, 2016-08-16:
Thanks,I'll try the Windoze test as soon as I get a chance.I was thinking about taking a look at Mono to see if this issue is on the "bug" or new features list and if anyone is working on addressing it.But I'll look at making the change in GRBL.cs as a quicker way to resolve on Linux.Burt

(#28)

smitjs, 2016-08-17:
Please tell me the connection between GRBL and the probe.
I am using Windows to GRBL on a Arduino Uno, and I read that A5 is the input on the GRBL.
Is this correct for your app?
Thank you, I will really like to try it.
Johan Smit

martin2250, 2016-08-17:
Hi Johan,
this is correct, just use A5. it has the internal pullup enabled so you won't need any additional hardware.
Anyways, just try it out in mid-air and touch some wires together as a first test before risking any hardware.
You're welcome to create new issues for questions, hijacking other issues just adds confusion.

smitjs, 2016-08-18:
Thank you Martin,

Sorry, I thought the title of the issue was correct.

My mistake.

Best Regards

Johan smit

 

  _____  

 

<https://github.com/martin2250/GrblHeightProbe2/issues/28#issuecomment-24049
1483>  it on GitHub, or mute
Pyy2a34ks5qg0nQgaJpZM4JjwL1>  the thread.
<https://github.com/notifications/beacon/AHylIlL5a2doGW4m6pJLek5tANCSZyIlks5
qg0nQgaJpZM4JjwL1.gif>

iw2fvo, 2017-01-07:
Hi to all in this forum,
I can control the cnc by this program by sendin connad like G= X10 etc. and so think the connection is OK.
I can not do anything more:  coul anyone help me sending step by step procedure to apply height probing to a gcode file please ? ( a sample gcde file is appreciate just to avoid additional problems ...)
I am leaving my mail address: iw2fvo@yahoo.co
Thanks for all 
regards,
Ambrogio

martin2250, 2017-01-07:
this is not a forum, please don't hijack other issues with new problems, instead open a new issue so other people with similar problems can find it more easily.
As for your problem, can you reproduce the steps shown in the video?

closed 2017-01-07 by martin2250

iw2fvo, 2017-01-08:
Thanks,I am sorry if I did post in the wrong place: give me a link where to post pls.I have grbl 1.1e on my win_7 PC. It is working well with my CNC.If I do launch the grblheightprobe2.exe (1270 KB ) I have problems.> connection on com 6 seems to be ok and I am able to move the CNC ( e.g. G= x 10 produce a correct displacement ) using the program.> GCode tab > open : If I do click on a GRBL file it opens a second windows called GCode Preview containing the drawing. >>  I am not able to go on : could you please let me have a step by step sequence  to continue ? I just want to apply z height correction for PCB milling.Thank Regards,Ambro