martin2250/GrblHeightProbe2 #29: Short Questions Thread

martin2250, 2016-08-17:
Since some people didn't want to create their own issues for short questions I've created this permanent issue. You can also use it for discussions on the topic.

momiccioli, 2016-08-21:
Where is the executable?  I downloaded the zip but there is no exe.

martin2250, 2016-08-21:
https://github.com/martin2250/GrblHeightProbe2/releases
this is also linked to from the readme and the youtube video.

momiccioli, 2016-08-21:
Found it and tried it.  Will this run with TinyG controller?  It connects but the commands don't get executed.  It would be great to have this and the Gcode sender all in one package.

martin2250, 2016-08-21:
No, only grbl is supported. a while ago I started working on an integrated sender/autoleveller (with support for marlin/tinyg) but I just don't have the time to work on it.

momiccioli, 2016-08-21:
> <reply+01080a2d984b1d7ada49c1684530c9c794019b518e504b7592cf0000000113d1941e9
> 2a169ce0a3c43b0@reply.github.com>
> <comment@noreply.github.com>

Martin,

Thanks for the update, I know how it is thank you.

<https://github.com/martin2250/GrblHeightProbe2/issues/29#issuecomment-24126
vmlzsOSks5qiHgegaJpZM4JmuLT> .

  
b","name":"GitHub"},"entity":{"external_key":"github/martin2250/GrblHeightPr
obe2","title":"martin2250/GrblHeightProbe2","subtitle":"GitHub
repository","main_image_url":"https://cloud.githubusercontent.com/assets/143
418/17495839/a5054eac-5d88-11e6-95fc-7290892c7bb5.png","avatar_image_url":"h
ttps://cloud.githubusercontent.com/assets/143418/15842166/7c72db34-2c0b-11e6
-9aed-b52498112777.png","action":{"name":"Open in
GitHub","url":"https://github.com/martin2250/GrblHeightProbe2"}},"updates":{
"snippets":[{"icon":"PERSON","message":"@martin2250 in #29: No, only grbl is
the time to work on it."}],"action":{"name":"View
Issue","url":"https://github.com/martin2250/GrblHeightProbe2/issues/29#issue
comment-241266145"}}}