martin2250/GrblHeightProbe2 #31: add option

ilie321, 2017-12-30:
hello 
can you add a option to get data automatic for map (height width) from gc code? is a good think and simplify the thinks. i already did for me but is a litle messy programming :) . so if you open first a gcode file you should have a button do get sizes auto like candle software for grbl has.  also i added a check to keep this sizes unmodified when you change grid dimension but modify number of points instead

martin2250, 2017-12-30:
Hi,
I already added this feature, though apparently before creating the latest release.
I like the Idea of having a button to set the correct size instead of doing it automatically. I'll add this and upload the next release soonish.

Thanks!
Martin

martin2250, 2017-12-30:
Wait, I just realized that this is not the thread for OpenCNCPilot. GrblHeightProbe is no longer supported. Did you mean OpenCNCPilot?
Anyways, I'll add it to OpenCNCPilot. If you are still using the old software you should switch anyhow :)

ilie321, 2017-12-30:
ok i will use pilot. i used the other one. thanks for the fast response. continue the good work

closed 2017-12-30 by martin2250